vshymanskyy/TinyGSM #764: BG96 and BG95-M3 board support and compatibility?

GavinBeattie, 2023-12-18:
[X] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[X] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (i.e., sending data over TCP/IP)
[X] Question or request for help

### What are you working with?

Modem: Quectel BG95-M3
Main processor board: Pico W

### Scenario, steps to reproduce
Hi the Quectel BG96 is by all accounts very similar to the BG95. I have the BG95-M3 as part of the [SixFab Pico Lite LTE](https://sixfab.com/product/sixfab-pico-lte/)

I'm currently gearing up for a switch from MicroPython (the main language the board supports) to Arduino. Would the BG96 board support cover the BG95-M3 and BG95*? If not, what would be required to pull it through? I appreciate there is probably a queue of board support requests, but this one seems fiendishly close. It would hopefully onboard a chunk of other hardware to the joy of TinyGSM.

our30K, 2024-03-21:
any progress in this ?

Humancell, 2024-04-08:
Hello,

I also just got a Sixfab Pico LTE and am about to do some testing.  @GavinBeattie and @our30K did either of you try using the BG96 driver?  Did it work?  What were the specific issues?

I'll do some testing and see what I see also, but was curious what issues you ran into.

Humancell, 2024-04-08:
Just as an update on my testing.  So far the BG96 drive is appearing to work.  There are some steps I had to take in advance to reset the modem and enable it, but then it did register.  There are some modem options that are not exposed through functions and I'm looking at those now.  Setting LTE/GPRS, CAT-M/NB-IoT/2G, and GPS functions.  I'm also looking at if I simply clone the BG96 driver and begin to hack it for the BG95 to add the missing functionality.   I'll keep you up to date.

GavinBeattie, 2024-04-09:
@Humancell, That sounds great. I've pivoted onto SIM7080G devices for now, but I would like to keep up with any progress you make. If you can share/post your precursor steps, I'll try and gather some support and maybe get back on it myself.

Humancell, 2024-04-29:
Ok, I've got my first experiment working if anyone else wants to give it a try.

I'm realizing that there really ought to be the BG95 driver written, but that is a future project. :-) 

For now, if got things working and welcome feedback. This is a simple asset tracking demo that has options to post via HTTP or MQTT.

I even designed a 3D printable enclosure for the board and some Qwiic sensors.

Let me know if it works for you, and what needs to be fixed!

https://github.com/Wovyn/sixfab-pico-lte-asset-tracker-example