vshymanskyy/TinyGSM #766: SIM7600SA support

CcKefa, 2024-01-04:
[ x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ x] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: SIMCOM SIM7600SA
Main processor board: ESP32
TinyGSM version: (0.11.7) 
Code: 
#define TINY_GSM_MODEM_SIM7600

#define SerialMon Serial

// Set serial for AT commands (to the module)
#define SerialAT Serial1

// #define TINY_GSM_DEBUG SerialMon

#include <TinyGsmClient.h>

// Module baud rate
uint32_t rate = 115200; // Set to 0 for Auto-Detect

void setup() {
    Serial.print("Setup comenzando");
  // Set console baud rate
  SerialMon.begin(115200);
  delay(6000);
}

void loop() {

  if (!rate) {
    SerialMon.println(F("***********************************************************"));
    SerialMon.println(F(" Module does not respond!"));
    SerialMon.println(F("   Check your Serial wiring"));
    SerialMon.println(F("   Check the module is correctly powered and turned on"));
    SerialMon.println(F("***********************************************************"));
    delay(30000L);
    return;
  }

  SerialAT.begin(rate, 18, 19);

  // Access AT commands from Serial Monitor
  SerialMon.println(F("***********************************************************"));
  SerialMon.println(F(" You can now send AT commands"));
  SerialMon.println(F(" Enter \"AT\" (without quotes), and you should see \"OK\""));
  SerialMon.println(F(" If it doesn't work, select \"Both NL & CR\" in Serial Monitor"));
  SerialMon.println(F("***********************************************************"));

  while(true) {
    if (SerialAT.available()) {
      SerialMon.write(SerialAT.read());
    }
    if (SerialMon.available()) {
      SerialAT.write(SerialMon.read());
    }
    delay(0);
  }
}

### Scenario, steps to reproduce
I'm trying to run the simple AT_Debug code and i'm unable to stablish connection to the module. I've tried everything that could make it work and no result so i'm assuming it's a compatibility issue.

### Expected result
Make the code work

### Actual result
You might have seen that these lines:
  if (!rate) {
    rate = TinyGsmAutoBaud(SerialAT);
  }
 Are missing from the code. This is because it crashes my program. The output is:

 ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

### Debug and AT command log

***********************************************************
 You can now send AT commands
 Enter "AT" (without quotes), and you should see "OK"
 If it doesn't work, select "Both NL & CR" in Serial Monitor
***********************************************************
---- Sent utf8 encoded message: "AT" ----

Veccoy, 2024-02-01:
I'm facing the same problem, did you solve it ?

Veccoy, 2024-02-06:
> Hi, i was talking with the supplier and it seems that my board was faulty. He has sent a new one but i haven't received it yet.

Hi, thank you for your answer. I figured out that the wiring was the culprit, two pins were connected to each other because of a bad welding...

----- Mail original -----

https://github.com/vshymanskyy/TinyGSM/issues/766#issuecomment-1926802619

CcKefa, 2024-02-06:
Hello, which pins did you connect? There is a lot of different configurations. Some say that with the Tx and Rx (and power of course) is enough.

https://github.com/vshymanskyy/TinyGSM/issues/766#issuecomment-1926802619

Veccoy, 2024-02-13:
Hi, I first connect the RX and TX of the modem to UART0, which I figured out later was a bad idea because of the intrinsic functions of the UART0

CcKefa, 2024-02-13:
Hello, thanks for your response. If you are using a ESP32 remember you have 3 serial ports to connect to. The new board just arrived from the manufacturer, let's hope I can make this one work lol.