vshymanskyy/TinyGSM #770: SIM7600X SSL/Secure Client?

nickdaria, 2024-01-15:
I have implemented TinyGSM w/ the 7600NA but unlike other chips, the 7600 lacks a TinyGsmClientSecure implementation in TinyGSM. Taking a look, it [appears it was started but not ever tackled](https://github.com/vshymanskyy/TinyGSM/blob/master/src/TinyGsmClientSIM7600.h). (see TODO w/ commented out TinyGsmClientSecure)

I had to build a very hacky AT command implementation to run alongside TinyGSM to make it work, but I would like to see this implemented down the road for future projects with the 7600 so everything can stay behind the TinyGSM layer.

Unfortunately, I do not currently have the time to familiarize myself with TinyGSM source and make a PR with this. If there is no progress on this when I have time later on, I will update this issue with any progress. That said, I would love for an active contributor to take a look if at all possible.

## What type of issues is this?
Request to implement TinyGsmClientSecure on 7600X

### What are you working with?
Modem: 7600NA
Main processor board: ESP32
TinyGSM version: 0.11.7

floBik, 2024-02-07:
Hi @nickdaria,

have a look at PR #767. I think this should be the solution you need.

nickdaria, 2024-02-07:
> Hi @nickdaria,
> 
> have a look at PR #767. I think this should be the solution you need.

That looks excellent! I will give that fork a shot tonight and report back. Thanks for the heads up!