vshymanskyy/TinyGSM #782: [ Question ] ESP32 ### unhandled:

RazorHex, 2024-02-13:
[x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[x] Question or request for help

### What are you working with?
Hello, i'm trying to connect SIM7600G with ESP32 but i don't understand the output and how to set up the debug mode.
I have tried the solution provided in [this comment](https://github.com/vshymanskyy/TinyGSM/issues/91#issuecomment-356024747) but i still not understand the output and where to find the documentation about the expected output.

**Modem**: SIM7600G
**Main processor board**: DOIT ESP32 DevKit v1
**TinyGSM version**: 0.11.7
**Code**: 
[image]

### Scenario, steps to reproduce

### Expected result

### Actual result
[image]

### Debug and AT command log

live-alchemy, 2024-02-23:
Hey @RazorHex, I have a similar issue. Are you willing to paste your code so I can try and see if I get the same output?

I also can't connect to the GSM Module and send AT commands.

I am using an ESP32 (ESP32 S3 DevKit-1) and SIMCom7600G.

RazorHex, 2024-02-23:
This is my current code, it work without the sim card and able to capture gnss output.
```cpp
#include "4G_handler.h"

const char *broker = "broker.hivemq.com"; // MQTT Credential

HardwareSerial SerialAT(2);

#ifdef DUMP_AT_COMMANDS
StreamDebugger debugger(SerialAT, SerialMon);
TinyGsm _modem(debugger);
#else
TinyGsm _modem(SerialAT);
#endif

// TinyGsm _modem(SerialAT);
TinyGsmClient client(_modem);

SIM7600G::SIM7600G() {}

boolean SIM7600G::_reset()
{
  return _modem.factoryDefault();
}

boolean SIM7600G::init(boolean reset)
{
  SerialAT.begin(defaultBaud, SERIAL_8N1, RX_GSM, TX_GSM, false);

  if (_modem.isGprsConnected())
  {
    if (_modem.enableGPS())
    {
      SerialMon.println("GPS Enabled.");
    }

    return _modem.isGprsConnected();
  }

  delay(6000);

  SerialMon.print("Setting Baud Rate to ");
  SerialMon.println(defaultBaud);
  _modem.setBaud(defaultBaud);
  // TinyGsmAutoBaud(SerialAT, GSM_AUTOBAUD_MIN, GSM_AUTOBAUD_MAX); // use for testing

  if (reset)
  {
    SerialMon.println("Reseting the module...");
    _reset();
  }

  delay(6000);

  SerialMon.println("SerialAT Ready!");

  SerialMon.println("Initializing SIM7600G...");
  _modem.init();
  delay(3000);

  SerialMon.print("Sim Card Status : ");
  SerialMon.println(_modem.getSimStatus());
  delay(3000);

  String modemInfo = _modem.getModemInfo();
  SerialMon.print("Modem Info : ");
  SerialMon.println(modemInfo);
  delay(3000);

  SerialMon.println("Connecting to GPRS...");
  if (!_modem.waitForNetwork(10000L))
  { // You may need lengthen this in poor service areas
    SerialMon.println(F(" [fail]"));
    SerialMon.println(F("************************"));
    SerialMon.println(F(" Is your sim card locked?"));
    SerialMon.println(F(" Do you have a good signal?"));
    SerialMon.println(F(" Is antenna attached?"));
    SerialMon.println(F(" Does the SIM card work with your phone?"));
    SerialMon.println(F("************************"));
    delay(10000);
  }
  _modem.gprsConnect(apn);

  if (_modem.waitResponse(10000L) != 1)
  {
    Serial.println("GPS FAIL");
    // Serial.println("SGPIO=0,4,1,1 false ");
  }
  if (_modem.enableGPS())
  {
    SerialMon.println("[ + ] GPS Enabled.");
  }
  else
  {
    SerialMon.println("[ - ] GPS Disabled.");
  }

  // SerialMon.println("Setting mode for LTE");
  // if (_modem.setNetworkMode(38) > 0)
  // {
  //   if (_modem.gprsConnect(apn) > 0)
  //   {
  //     SerialMon.println("Connected!");
  //   }
  //   break;
  // }
  // else
  // {
  //   SerialMon.println("Failed!");
  // }

  // bool conn = _modem.isGprsConnected();
  return _modem.isGprsConnected();
  // SerialMon.println((conn) ? "Connected" : "Failed");
}

String SIM7600G::getGPS()
{
  if (SerialAT.available() == 0)
  {
    SerialAT.begin(defaultBaud, SERIAL_8N1, RX_GSM, TX_GSM, false);
  }

  if (_modem.enableGPS() && !gps_enabled)
  {
    gps_enabled = true;
    SerialMon.println("GPS Enabled.");
  }

  float lat, lon;

  _modem.getGPS(&lat, &lon);
  SerialMon.println(lat);
  SerialMon.println(lon);

  return _modem.getGPSraw();
}
```

This is the defined variable
```cpp
#define TINY_GSM_MODEM_SIM7600
#define SerialMon Serial
#define TINY_GSM_DEBUG SerialMon
#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false
#define GSM_AUTOBAUD_MIN 9600
#define GSM_AUTOBAUD_MAX 115200
#define DUMP_AT_COMMANDS

#define TINY_GSM_TEST_GPRS true
#define TINY_GSM_TEST_TCP true
#define TINY_GSM_TEST_GPS true

#define RX_GSM 16
#define TX_GSM 17
#define defaultBaud 115200

#include <TinyGSM.h>
#include <TinyGsmClient.h>
#include <HardwareSerial.h>
#include <StreamDebugger.h>
```

live-alchemy, 2024-02-23:
Hm ok, I'm afraid I'm not totally clear which problem you're facing or what you want to accomplish. Apologies.

RazorHex, 2024-02-26:
Yeah same here, i think its related to the IMEI

AapoTahkola, 2024-05-29:
I promptly reflashed global firmware to make SIM900A SIM900. 
I had garbage on the io line until I changed circuit so that TX from esp32 goes to 5V RX on SIM900A. It works fine I can do firmware update on GPRS line to esp32.

closed 2025-08-08 by AhdaAI