vshymanskyy/TinyGSM #786: SSL support to SIM7600

ramonpaolo, 2024-03-29:
[x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[x] Request to support a new module
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: SIM 7600
Main processor board: Uno
TinyGSM version: 0.11.7
Code: 

The SIM 7600 not have yet a support to SSL connections!

floBik, 2024-04-02:
There is already a pull request (#767 ) from @Matt-Stedman and myself to add this feature, awaiting approval. 
Perhaps @vshymanskyy or @SRGDamia1 could take a look at this pr.

Gspohu, 2024-07-17:
Any news on it ?

don41382, 2024-09-07:
2 month past again. Any update when this will be merged?

floBik, 2024-09-09:
I recently created a new PR(#808 ) with a bump to the current TinyGSM version and some refactoring.

Pakingster, 2024-09-20:
> I recently created a new PR(#808 ) with a bump to the current TinyGSM version and some refactoring.

Is MQTT TLS connection supported, what a chance you have any example for this? 
Thanks

Matt-Stedman, 2026-01-12:
> > I recently created a new PR([#808](https://github.com/vshymanskyy/TinyGSM/pull/808) ) with a bump to the current TinyGSM version and some refactoring.
> 
> Is MQTT TLS connection supported, what a chance you have any example for this? Thanks

Yes, it is supported - I'll see about making an example shortly. (I've tested with AWS IoT Core).

However, seriously please @vshymanskyy this has been open and ready for nearly 2 years, @floBik and I worked hard on it, I believe it's ready to merge, and the community really need these features - if you have feedback please let us know otherwise I don't understand what the hold up is?

apeman76, 2026-02-04:
bump @vshymanskyy