vshymanskyy/TinyGSM #796: `write()` method fails with frequency with new version `v0.12.0`

Decezaris, 2024-05-31:
## What type of issues is this?

[ ] Request to support a new module
[ ] Bug or problem compiling the library
[ ] Bug or issue with library functionality (ie, sending data over TCP/IP)
[x] Question or request for help

### What are you working with?

Modem: A7670SA
Main processor board: ESP32
TinyGSM version: v0.12.0

I am including `SIM7600` in my code and my `rx buffer` is setted to 1024 Bytes, I updated the library version all works fine at the beggining but with a while I noticed a lot of write fail errors compared to `v0.11.7` version as shown bellow:

[image]

SRGDamia1, 2024-06-03:
I'll look into this. Can you give me more information about the code you're running?

Decezaris, 2024-06-04:
I'll try to find the conditions again with debug mode active, It's not easy to reproduce, what I can antecipate is that the signal conditions it was very good (31/31). About my code it's using tinyGSM, Govorox sslClient and ArduinoMQTT for connection with AWS IoT Core. Bellow some new occurrences of this error.

```
10:41:42.190 -> NTP sync serv_0: 0 serv_1: 255 serv_2: 255 serv_3: 255
10:41:42.227 -> Successful to sync.
10:41:42.227 -> dt from modem: 24/06/02,13:41:41+00
10:41:42.227 -> unix timestamp: 1717324901
10:41:42.227 -> Try connection with AWS... Signal strength: 100
10:41:42.227 -> -FLASH: read from file /last_rtc_update.txt
10:41:42.227 -> [ntpRightShared]: 1
10:41:42.227 -> Reconnection
10:41:42.227 -> error ssl_client: ERROR - Generic error. Connected with final certificates
10:41:42.280 -> [ 71717][D][SSLClient.cpp:512] setCACert(): Set root CA
10:41:42.280 -> [ 71725][D][SSLClient.cpp:547] setCertificate(): Set client CA
10:41:42.280 -> [ 71730][D][SSLClient.cpp:562] setPrivateKey(): Set client PK
10:41:44.001 -> [ 73381][D][ssl__client.cpp:724] perform_ssl_handshake(): calling mbedtls_ssl_handshake with ssl_ctx address 0x3ffbc5a4
10:41:48.547 -> [ 77963][D][ssl__client.cpp:745] perform_ssl_handshake(): Protocol is TLSv1.2 Ciphersuite is TLS-ECDHE-RSA-WITH-AES-128-GCM-SHA256
10:41:48.547 -> [ 77975][D][ssl__client.cpp:748] perform_ssl_handshake(): Record expansion is 29
10:41:48.547 -> [ 77982][D][ssl__client.cpp:254] cleanup(): Free internal heap after TLS 181960
10:41:48.547 -> [ 77989][I][SSLClient.cpp:248] connect(): SSL connection established
10:41:48.547 -> [ 77996][D][SSLClient.cpp:380] write(): Sending data to SSL connection...
10:41:49.175 -> Signal strength: 100
10:41:49.175 -> isSessionPresent: 1
10:41:49.717 -> Set retry time to 10 seconds
10:41:49.717 -> FIFO size before publish: 2
10:41:49.717 -> [ 79157][D][SSLClient.cpp:380] write(): Sending data to SSL connection...
10:41:49.717 -> [ 79188][D][SSLClient.cpp:380] write(): Sending data to SSL connection...
10:41:50.921 -> [ 80242][E][ssl__client.cpp:193] client_net_send(): write failed
10:41:50.921 -> [ 80247][E][ssl__client.cpp:46] _handle_error(): [send_ssl_data():901]: (-78) Failed to send data - underlying network layer error
10:41:50.921 -> [ 80259][E][SSLClient.cpp:384] write(): Error sending data to SSL connection. Stopping SSLClient...
10:41:50.921 -> [ 80268][D][SSLClient.cpp:97] stop(): Stopping ssl client
10:41:50.921 -> [ 80273][D][ssl__client.cpp:810] stop_ssl_socket(): Cleaning SSL connection.
10:41:50.921 -> [ 80280][D][ssl__client.cpp:814] stop_ssl_socket(): Stopping SSL client. Current client pointer address: 0x3ffc2ba8
10:41:50.921 -> [ 80297][D][ssl__client.cpp:819] stop_ssl_socket(): Freeing CA cert. Current ca_cert address: 0x3ffbcb74
10:41:50.921 -> [ 80306][D][ssl__client.cpp:827] stop_ssl_socket(): Freeing client cert and client key. Current client_cert address: 0x3ffbcccc, client_key address: 0x3ffbce24
10:41:50.921 -> [ 80320][D][ssl__client.cpp:835] stop_ssl_socket(): Freeing SSL context. Current ssl_ctx address: 0x3ffbc5a4
10:41:50.921 -> [ 80330][D][ssl__client.cpp:838] stop_ssl_socket(): Freeing SSL config. Current ssl_conf address: 0x3ffbc7c4
10:41:50.921 -> [ 80340][D][ssl__client.cpp:841] stop_ssl_socket(): Freeing DRBG context. Current drbg_ctx address: 0x3ffbc8ac
10:41:50.921 -> [ 80350][D][ssl__client.cpp:844] stop_ssl_socket(): Freeing entropy context. Current entropy_ctx address: 0x3ffbc8fc
10:41:50.921 -> [ 80360][D][ssl__client.cpp:847] stop_ssl_socket(): Finished cleaning SSL connection.
10:41:50.921 -> [ 80368][D][SSLClient.cpp:97] stop(): Stopping ssl client
10:41:50.921 -> [ 80373][D][ssl__client.cpp:810] stop_ssl_socket(): Cleaning SSL connection.
10:41:50.921 -> [ 80380][D][ssl__client.cpp:814] stop_ssl_socket(): Stopping SSL client. Current client pointer address: 0x3ffc2ba8
10:41:50.986 -> [ 80447][D][ssl__client.cpp:835] stop_ssl_socket(): Freeing SSL context. Current ssl_ctx address: 0x3ffbc5a4
10:41:51.057 -> [ 80457][D][ssl__client.cpp:838] stop_ssl_socket(): Freeing SSL config. Current ssl_conf address: 0x3ffbc7c4
10:41:51.057 -> [ 80467][D][ssl__client.cpp:841] stop_ssl_socket(): Freeing DRBG context. Current drbg_ctx address: 0x3ffbc8ac
10:41:51.057 -> [ 80477][D][ssl__client.cpp:844] stop_ssl_socket(): Freeing entropy context. Current entropy_ctx address: 0x3ffbc8fc
10:41:51.057 -> [ 80487][D][ssl__client.cpp:847] stop_ssl_socket(): Finished cleaning SSL connection.
10:41:51.057 -> FIFO size after try to publish: 2
10:41:52.442 -> NTP sync serv_0: 0 serv_1: 255 serv_2: 255 serv_3: 255
10:41:52.442 -> Successful to sync.
10:41:52.442 -> dt from modem: 24/06/02,13:41:52+00
10:41:52.442 -> unix timestamp: 1717324912
10:41:52.442 -> Try connection with AWS... Signal strength: 100
10:41:52.442 -> -FLASH: read from file /last_rtc_update.txt
10:41:52.442 -> [ntpRightShared]: 1
10:41:52.442 -> Reconnection
```

SherifMEldeeb, 2024-09-04:
Greetings @Decezaris, 
I guess I have the same issue as the ssl connection fails randomly with different error messages.
Did you have any progress with this?

SherifMEldeeb, 2026-01-09:
> Hello everyone!!
> 
> I have the same situation here.
> 
> [1155658][E][ssl__client.cpp:192] client_net_send(): write failed
> [1155664][E][ssl__client.cpp:45] _handle_error(): [send_ssl_data():919]: (-78) Failed to send data - underlying network layer error
> [1155676][E][SSLClient.cpp:384] write(): Error sending data to SSL connection. Stopping SSLClient...
> 
> This is very painfull to my project!

Yeah this driver either it works or it gets very painful and frustrating.