vshymanskyy/TinyGSM #797: A7672E with ESP32 connects to APN, but doesn't connect to MQTT Broker

lg007, 2024-06-07:
[x] I have read the Troubleshooting section of the ReadMe

## What type of issues is this?

[ ] Request to support a new module
    
[ ] Bug or problem compiling the library
[x] Bug or issue with library functionality (ie, sending data over TCP/IP)
[ ] Question or request for help

### What are you working with?

Modem: A7672E
Main processor board: ESP32 devkit
TinyGSM version: 12.0
Code: The MQTTClient example code (with own APN set and with the modification that is needed for the ESP32)

### Scenario, steps to reproduce
Just try the MQTTClient example with the modification that is needed for the ESP32: 
`SerialAT.begin(115200, SERIAL_8N1, 16, 17, false) // the modem is connected to pins 16 and 17`

### Expected result
I expect being able to connect to the MQTT Broker

### Actual result
Connection to the network is OK (the LED of the modem blinks fast). The connection to the MQTT Broker fails.

### Debug and AT command log
If I reproduce the same circuit with another module (a SIM800L), it works and connects to the MQTT Broker (with the same APN). Of course I change the `#define` for the modem that I use.

lg007, 2024-06-11:
An interesting fact: if I change the MODEM define to the following:
`#define TINY_GSM_MODEM_SIM7600`

instead of this:
`#define TINY_GSM_MODEM_A7672X`
then everything works!

Has anyone any idea?

SRGDamia1, 2024-06-12:
Hmm. A lot of the SIMCom modules are very similar and have more than one way to connect. I wonder if it's something to do with the "application" stack vs the TCP stack. That's why there are two versions of the SIM7000.