vshymanskyy/TinyGSM #805: ESP32 Webserver

Gradhoc-aeimil, 2024-08-22:
Hi, I'm trying to setup a webserver through a module simcom 7080, but this is not supported right now in the library. Is there any plan on the future to include this feature?

Thanks.

CF20852, 2024-09-20:
I have the same question, but I'm using a SIM7000G.

elvedin13, 2025-02-05:
I have done it with SIM7600 it should probably work on your modem.
Add to your simcom modem.h file:

```
 public:
// Additional methods for SIM7600 TCP Server mode
    bool startServer(uint16_t port);
    bool stopServer();
    bool checkForIncomingConnection();
    String readIncomingData();
    void sendResponse(const String &data);
```

add to your modem .cpp file:

```
#include "TinyGsmClientSIM7600.h"

int activeLinkID = -1;  // Store active client link number
String clientIP = "";   // Store client IP
int clientPort = 0;     // Store client port
// Start TCP server on specified port
bool TinyGsmSim7600::startServer(uint16_t port) {
    sendAT(GF("+SERVERSTART="), port, GF(",0")); // Start TCP server
    if (waitResponse(GF("OK")) != 1) {
        Serial.println("Failed to start TCP server!");
        return false;
    }

    Serial.println("TCP Server started successfully on port " + String(port));
    return true;
}

// Stop the TCP server
bool TinyGsmSim7600::stopServer() {
    sendAT(GF("+SERVERSTOP=0"));  // Stop TCP server
    if (waitResponse(GF("OK")) != 1) {
        Serial.println("Failed to stop TCP server!");
        return false;
    }

    sendAT(GF("+NETCLOSE"));  // Close network
    return waitResponse(GF("OK")) == 1;
}

// Read incoming HTTP request properly
bool TinyGsmSim7600::checkForIncomingConnection() {
    if (stream.available()) {
        String response = stream.readStringUntil('\n');  
        if (response.startsWith("+CLIENT:")) {
            Serial.println("Client connected: " + response);

            // Extract Link ID from "+CLIENT: <link_id>,..."
            int firstComma = response.indexOf(',');
            if (firstComma > 0) {
                activeLinkID = response.substring(8, firstComma).toInt();  // Extract first number
                Serial.println("Extracted Active Link ID: " + String(activeLinkID));
            }
            return true;
        }
    }
    return false;
}

// Read incoming HTTP request
String TinyGsmSim7600::readIncomingData() {
    String request = "";
    sendAT(GF("AT+CIPRXGET=2,0,200")); // Read available data

    while (stream.available()) {
[... 129 lines cut ...]
  Serial.println("Server started successfully. Waiting for connections...");
}

void loop() {
  // Check for an incoming client connection
  if (modem.checkForIncomingConnection()) {
    Serial.println("Client connected. Reading request...");

    // Read the HTTP request from client
    String request = modem.readIncomingData();
    Serial.println("Request Received:");
    Serial.println(request);

    // Create HTTP response
    String response = "HTTP/1.1 200 OK\r\n"
                      "Content-Type: text/html\r\n\r\n"
                      "<html><body><h1>Hello from ESP32 + SIM7600!</h1></body></html>";

    // Send the response back to the client
    modem.sendResponse(response);

    Serial.println("Response sent.");
  }
}
```