vshymanskyy/TinyGSM #812: Assigning dynamic topics for MQTT

yilmazyurdakul, 2024-11-06:
Main processor board: 
TinyGSM version: 

Thanks. I can provide additional info, debug or code block after preparing it, i need to make a simpler project to do it.

mirhamza708, 2024-11-11:
Please first go through the steps of how to put your code in github so other people can read it. Your code here is unreadable and for me at least, I wont be able to help.

digicyc, 2024-11-15:
One question, is it possible you're trying to connect to the broker more than once?

yilmazyurdakul, 2024-11-17:
I solved the issue. Thanks for your time. i will share the snippet. 

`  // Initialize topicDoor
  static char topicDoorBuffer[34];
  snprintf(topicDoorBuffer, sizeof(topicDoorBuffer), "FleetLink/devices/%s/door", internalId);
  topicDoor = topicDoorBuffer;`
  
  after resetting the board, code started to work. Also i realized that i was having some timing issue about reading spiffs. I am not sure about that problem but its just a theory. 
  
  The good thing is "problem solved!"
  
  Thanks