vshymanskyy/TinyGSM #816: Several issues with TinyGsmClient

TheSnowFall, 2024-12-23:
Today I updated ESP32 core (3.1.0) from the board manager then the following issues are showing while compiling code regarding TinyGSM.

Code snippet:

```
#include <Arduino.h>
#include <string.h>
#include <stdlib.h>

#include <WiFiManager.h>
#include <PubSubClient.h>
#include <TinyGsmClient.h>

#define GSM_PIN ""

#define SerialGeneric Serial
#define SerialAT Serial1
                                            // #define DUMP_AT_COMMANDS
#define TINY_GSM_DEBUG SerialGeneric

#ifdef DUMP_AT_COMMANDS
#include <StreamDebugger.h>
StreamDebugger debugger(SerialAT, SerialGeneric);
TinyGsm modem(debugger);
#else
TinyGsm modem(SerialAT);
#endif

#include "utilities.h"

TinyGsmClient client_gprs(modem);
PubSubClient mqtt_gprs(client_gprs);

```

I just shared only this portion because it is not being used anywhere else. You may also find those in the mqtt_client example of TinyGSM. This code was working 3 days ago last time I worked. Today I just started/turned one arduino IDE and the IDE asked me ESP32 core update. After giving the update I started compiling. Then the error came out. Note that I did not change anything in the code. The error is below:
```
C:\Users\ERFAN-HOME-PC\Documents\GitHub\SFBD_My_Playground\SFBD_Gateway\SFBD_Gateway.ino:162:15: error: cannot declare variable 'client_gprs' to be of abstract type 'TinyGsmSim800::GsmClientSim800'
  162 | TinyGsmClient client_gprs(modem);
      |               ^~~~~~~~~~~
In file included from c:\Users\ERFAN-HOME-PC\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClient.h:13,
                 from C:\Users\ERFAN-HOME-PC\Documents\GitHub\SFBD_My_Playground\SFBD_Gateway\SFBD_Gateway.ino:14:
c:\Users\ERFAN-HOME-PC\Documents\Arduino\libraries\TinyGSM\src/TinyGsmClientSIM800.h:86:9: note:   because the following virtual functions are pure within 'TinyGsmSim800::GsmClientSim800':
   86 |   class GsmClientSim800 : public GsmClient {
      |         ^~~~~~~~~~~~~~~
In file included from C:\Users\ERFAN-HOME-PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0\cores\esp32/Arduino.h:197,
                 from C:\Users\ERFAN-HOME-PC\Documents\GitHub\SFBD_My_Playground\SFBD_Gateway\SFBD_Gateway.ino:8:
C:\Users\ERFAN-HOME-PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0\cores\esp32/Client.h:29:15: note:     'virtual int Client::connect(IPAddress, uint16_t, int32_t)'
   29 |   virtual int connect(IPAddress ip, uint16_t port, int32_t timeout) = 0;
      |               ^~~~~~~
C:\Users\ERFAN-HOME-PC\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.1.0\cores\esp32/Client.h:31:15: note:     'virtual int Client::connect(const char*, uint16_t, int32_t)'
   31 |   virtual int connect(const char *host, uint16_t port, int32_t timeout) = 0;
```

Has anyone solved it? I really appreciate any help you can provide.

TheSnowFall, 2024-12-23:
[Update] I reverted back to ESP32 3.0.7 from 3.1.0 and it is working fine.

aXinfor, 2024-12-27:
I updated ESP32 core (3.1.0) and it has the same error.
I have had to revert to version 3.0.7 to get it working properly again.
I haven't found a solution yet!
Any help?

mnltake, 2025-01-08:
https://github.com/espressif/arduino-esp32/pull/10776
This merged version built successfully
For PlatformIO, try specifying in the platformio.ini file
```
platform = https://github.com/platformio/platform-espressif32.git#master
```