vshymanskyy/TinyGSM #825: TinyGSM and generalized ESP-NETIF interfaces

johanneswen99, 2025-03-12:
I have TinyGSM successfully working to connect my Lilygo T-SIM 70x0g boards using simple client connections. However, I would like to establish generalized connections using the ESP IDF, e.g. an SSL connection using ESP-TLS. For this the TinyGSM module would need to create a ESP-NETIF network interface for PPP transmission. Such a generalized interface would extend the connectivity capabilities of TinyGSM immensely. 

Does anybody have any hints, suggestions, or experience how to create such an interface?

calvino80, 2025-04-01:
Hi,
Is this what you are describing?  Mobizt has examples in the FirebaseClient github:

example using NetworkClientSecure and DefaultPPPNetwork:

examples/App/NetworkInterfaces/DefaultNetworks/DefaultPPPNetwork  

and 

example using TinyGsmClient and ESP_SSLClient:

examples/App/NetworkInterfaces/GSMNetwork

johanneswen99, 2025-04-01:
Thanks, that looks good. I will have a more detailed look. For the moment I was resorting to the SIM 7xxx SSL client which is also neat because it runs entirely on the modem, but it's of course very hardware specific, so I'm still interested.

closed 2025-04-01 by johanneswen99

calvino80, 2025-04-01:
Sure, it's not written using the ESP-IDF but it does use the ESP32 to manage the SSL layer.