vshymanskyy/TinyGSM #827: SIM7600E   bad work tcip after ussd send

yuranikspb, 2025-03-31:
I have a device, works with mqtt.
Works quite well until send a Ussd balance request. After that, randomly connection disconnect tcp ip begins.
in my  log i changed mux for "1" for testing but no result.

```
12:12:47.797 -> AT+CUSD=1,"*100#"
12:12:47.832 -> 
12:12:47.832 -> OK
12:12:48.145 -> 
12:12:48.145 -> +CGPS: 0
12:12:50.343 -> 
12:12:50.343 -> +CUSD: 0,"0032003500200440002E000A041C043504330430041A043B0438043A04350440003A00200440043E0437044B04330440044B044800200434043E0020003100300030003000300440002E00200435043604350434043D04350432043D043E002C00200033003500200440002F0434002800310038002B0029003A002A0038003000310023",72
```

At the same time command 
```
 +CIPCLOSE: 0,1,0,0,0,0,0,0,0,0  look good
   OK
```
but:
```

AT+CIPSEND=1,43
+CIPERROR: 4
AT+NETOPEN?
 +NETOPEN: 1
OK
AT+IPADDR
+IPADDR: 10.19.82.50
AT+CIPRXGET=4,1
+CIPRXGET: 4,1,0
OK
AT+CIPCLOSE?
+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0
OK
 === MQTT NOT CONNECTED ===
 +CIPCLOSE: 1,4
 ERROR
 AT+CIPRXGET=1
 OK
AT+CIPOPEN=1,"TCP","*******.com",1883
 OK
 +CIPOPEN: 1,0
AT+CIPSEND=1,50

```
if restart modem work  fine until next ussd request. if not restart 90% mqtt commands not work/

yuranikspb, 2025-03-31:
if AT+CIPCLOSE=0  before  ussd 

it helps, but only before Ussd if after then it doesn't help, it's magic :(

droidblastnz, 2025-04-30:
> Works quite well until send a Ussd balance request

Same issue "works quite well until send a Ussd balance request" am I correct that you are requesting balance credit? 

 I have a function that called send BAL to 2424 with my network operator and the SIM7600 also is MQTT back to Home Assistant. It works well for so long and then I loose connection same as you. 

MQTT is using Data/TCP and Credit Balance text is using Cellular.  I have a routine that request the credit balance every 24hrs and then seems to be disconnecting the data TCP connection although my code tries to reconnect. The only way to resolve is hard reboot. 

So you mention closing the socket with AT+CIPCLOSE=0 does this resolve your issue? I think we need to close the TCP connection the request credit and the reopen the connection with AT+CIPOPEN. When my SIm7600 registers it gets a dynamic IP address with the AT+NETOPEN so perhaps the commands are AT+NETCLOSE and AT+NETOPEN? My logs on boot up which is the same as a hard reset.

So if you do a AT+NETCLOSE, request the credit balance SMS and then AT+NETOPEN AT+CIPOPEN=0,"TCP","mydomainame.com",1883` I wonder if this will resolve?

Or AT+CIPCLOSE=0 then AT+NETCLOSE send credit balance SMS then AT+NETOPEN AT+CIPOPEN=0,"TCP","mydomainame.com",1883`

`OK
Network state is: AT+NETOPEN?

+NETOPEN: 1

OK
AT+IPADDR

+IPADDR: xx.xx.xx.xx

OK
1
AT+NETOPEN?

+NETOPEN: 1

OK
AT+IPADDR

+IPADDR: xx.xx.xx.xx

OK
Connecting to mydomainame.com
AT+CIPRXGET=4,0

+CIPRXGET: 4,0,0

OK
AT+CIPCLOSE?

+CIPCLOSE: 0,0,0,0,0,0,0,0,0,0

OK
AT+CIPCLOSE=0

+CIPCLOSE: 0,4

ERROR
AT+CIPRXGET=1

OK
AT+CIPOPEN=0,"TCP","mydomainame.com",1883`

@

`void Refresh_Balance() {
    String smsNumber = "2424";                // Skinny is the Operator
    String smsMessage2 = "BAL";               // Request the balance of the account
    String command = "+CMGS=\"" + smsNumber + "\"";

    // Discard any leftover data in the modem buffer more thoroughly
    while (SerialAT.available() > 0) {
        SerialAT.read();
        delay(10);  // Slight delay to ensure buffer empties fully
    }
    delay(100);  // Additional delay to ensure buffer is clear added 10/02/2025

    // Send the AT+CMGS command to start the SMS sending process
    modem.sendAT(command);

    // Wait for the modem to respond with the '>' prompt
    unsigned long start = millis();
    bool promptReceived = false;
    while (millis() - start < 5000) {  // Increased timeout to 5000 ms for '>' added 10/02/2025
        if (SerialAT.available()) {
            char c = SerialAT.read();
            if (c == '>') {
                promptReceived = true;
                break;
            }
        }
    }

    if (!promptReceived) {  // If '>' not received within timeout
        pDBGln("Error: Modem did not prompt for message input.");
        return;
    }

    // Send the actual SMS message
    SerialAT.print(smsMessage2);

    // Wait a moment before sending the Ctrl+Z (ASCII 26) to finalize the message
    delay(500);  // Slight delay before termination character

    // Send the Ctrl+Z character to complete the SMS
    SerialAT.write(26);

    // Increase wait time after sending the message to handle network delays
    if (modem.waitResponse(10000L) == 1) {  // Increased timeout to 10000 ms for response 
        pDBGln("+CMGS Sent daily balance request!");
    } else {
        pDBGln("Error: Failed to send balance request.");
    }

    // Wait for the balance response SMS
    delay(10000);  // Wait for the response SMS to arrive

    // Read the SMS response
    String response = "";
    while (SerialAT.available()) {
        char c = SerialAT.read();
        response += c;
    }

    // Extract the balance value from the response
    String balance = "";
    bool foundBalance = false;
    for (int i = 0; i < response.length(); i++) {
        if (response[i] == '$' || (response[i] >= '0' && response[i] <= '9')) {
            foundBalance = true;
        }
        if (foundBalance && (response[i] == '.' || (response[i] >= '0' && response[i] <= '9'))) {
            balance += response[i];
        }
        if (foundBalance && response[i] == ' ') {
            break;
        }
    }

    // Print the extracted balance
    if (balance.length() > 0) {
        pDBGln("Extracted Balance: " + balance);
    } else {
        pDBGln("Error: Could not extract balance from response.");
    }
}`

droidblastnz, 2025-04-30:
Issue - T-sim7600 is connected to MQTT over 4G networks. When the code send BAL to 2424 the network operator sends the credit balance back. A lot of the time this works well but a lot of the time MQTT becomes disconnected and doesnt reconnectSIM7600 is switching network modes temporarily (e.g., SMS interrupts data session).

Sending or receiving SMS, especially via AT commands, can briefly disrupt the PDP (Packet Data Protocol) context, affecting MQTT.
PDP Context Deactivation or Network Drop

The module might lose its IP connection due to weak signal, network handover, or an operator-initiated drop, especially when juggling SMS + data.

MQTT client not retrying properly.  After a connection drop, your code may not reinitiate the MQTT connection correctly, or may fail silently if AT command responses aren’t handled properly.

SIM7600 firmware bugs or APN issues.  Some firmware versions have bugs with concurrent SMS + TCP sessions. Incorrect APN settings can also lead to unstable connections.

Resolve perhaps.
Add code that watches for disconnection and automatically reinitializes the MQTT session:
Periodically check with AT+CMQTTSTATUS or track +CIPCLOSE messages.
If disconnected, tear down and re-init:

`AT+CMQTTDISC=0,60
AT+CMQTTREL=0
AT+CGATT=1
AT+CGDCONT=1,"IP","your.apn"
AT+CMQTTSTART`

Separate SMS and MQTT Timing
Avoid sending SMS or parsing it while MQTT is active or during critical publish moments. Try to delay SMS sends or reads when MQTT is connected and active.
Ensure PDP Context is Stable
After sending SMS or on any error, verify network and IP are still up:

`AT+CGATT?           // Check if attached to network
AT+CGACT?           // Check if PDP context is active
AT+CGDCONT?         // Check IP config`

Full Recovery Logic
AT+CMQTTDISC=0,60       // Disconnect if possible
AT+CMQTTREL=0             // Release MQTT client
AT+CMQTTSTOP              // Stop MQTT service
AT+CGATT=1                   // Ensure attached
AT+CGDCONT=1,"IP","APN"
AT+CMQTTSTART
AT+CMQTTACCQ=0,"clientID"
AT+CMQTTCONNECT=0,"tcp://broker:port",keepalive,cleanSession

yuranikspb, 2025-05-05:
> So if you do a AT+NETCLOSE, request the credit balance SMS and then AT+NETOPEN AT+CIPOPEN=0,"TCP","mydomainame.com",1883` I wonder if this will resolve?
> 
> Or AT+CIPCLOSE=0 then AT+NETCLOSE send credit balance SMS then AT+NETOPEN AT+CIPOPEN=0,"TCP","mydomainame.com",1883`
> 
> From the manual - SIM7500_SIM7600 Series_AT Command Manual_V3.00 AT+NETCLOSE is used to stop socket service by deactivating PDP context. It also can close all the opened socket connections when you didn’t close these connections by AT+CIPCLOSE

what i do by timer:
`   modem.sendAT(GF("+CIPCLOSE=0")); 
    modem.waitResponse();
    ussd_balance = modem.sendUSSD(ussd_balance_cmd);
    DBG("Balance (USSD):", ussd_balance);`
 then 
in loop
```
 if (!modem.gprsConnect(apn.c_str(), gprsUser.c_str(), gprsPass.c_str())) {
      SerialMon.println(" fail, wait for 5 sec");
      delay(5000);
    }
    if (modem.isGprsConnected()) {
      SerialMon.println("GPRS reconnected");
      bool res = modem.isGprsConnected();
      DBG("GPRS status:", res ? "connected" : "not connected");
if (!mqtt.connected()) {
    SerialMon.println("=== MQTT NOT CONNECTED ===");
    // Reconnect every 10 seconds
    uint32_t t = millis();
    if (t - lastReconnectAttempt > 10000L) {
      lastReconnectAttempt = t;
      if (mqttConnect()) { lastReconnectAttempt = 0; }
    }
}
```
modem reconnect after ussd about 2 minutes/  
maybe i should try add AT+NETCLOSE after  CIPCLOSE

droidblastnz, 2025-05-05:
> what i do by timer: ` modem.sendAT(GF("+CIPCLOSE=0")); modem.waitResponse(); ussd_balance = modem.sendUSSD(ussd_balance_cmd); DBG("Balance (USSD):", ussd_balance);` then in loop
>
> ```
>  if [...]

Also posted  [here](https://github.com/Xinyuan-LilyGO/T-SIM7600X/issues/127) to get assistance as it seems to be a data socket connection then cellular sms and sometimes not reconnect.

droidblastnz, 2025-05-05:
> I have a device, works with mqtt. Works quite well until send a Ussd balance request. After that, randomly connection disconnect tcp ip begins. in my log i changed mux for "1" for testing but no [...]

Does your code work for a few times or fails immediately?  In my case it can work for days or weeks then disconnects. Generally I know when it is going to disconnect because the topic/balance returns 2424 not the actual credit value.

```
topic/balance 1.65 (good)
topic/balance 2424 (bad goes offline)
```

In my case the network operator you send 2424 BAL to then get a sms text with the balance back.

yuranikspb, 2025-05-05:
after add  `AT+NETCLOSE after CIPCLOSE`   before ussd, reconnect is very fast(about a 2-4 sec). 
Now I'll watch for it  for a while.

droidblastnz, 2025-05-06:
> after add `AT+NETCLOSE after CIPCLOSE` before ussd, reconnect is very fast(about a 2-4 sec). It turns out to be a good practice to close the socket (AT+CIPCLOSE) and the tcp-ip service (AT+NETCLOSE) before ussd or sms
> 
> Now I'll watch for it for a while.

Can you post you block of code not just the AT commands please?

droidblastnz, 2025-05-19:
> after add `AT+NETCLOSE after CIPCLOSE` before ussd, reconnect is very fast(about a 2-4 sec). It turns out to be a good practice to close the socket (AT+CIPCLOSE) and the tcp-ip service (AT+NETCLOSE) before ussd or sms
> 
> Now I'll watch for it for a while.

Any progress?

yuranikspb, 2025-05-22:
```
modem.sendAT(GF("+CIPCLOSE=0"));  ///socket close
    modem.waitResponse();
    modem.gprsDisconnect();  //tcp-ip close
    ussd_balance = modem.sendUSSD(ussd_balance_cmd);
    DBG("Balance (USSD):", ussd_balance);
```
and next in sketch

```
 if (!modem.gprsConnect(apn.c_str(), gprsUser.c_str(), gprsPass.c_str())) {
      SerialMon.println(" fail, wait for 5 sec");
      delay(5000);
    }
```
and then connect to mqtt